distatus/battery #1: Doesn't work on my Ubuntu installation

TheHippo, 2016-05-14:
`Uname`:

    $ uname -a
    Linux hippo 3.13.0-85-generic #129-Ubuntu SMP Thu Mar 17 20:50:15 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

`lsb-release`:

    $ lsb_release -a
    LSB Version:	core-2.0-amd64:core-2.0-noarch:core-3.0-amd64:core-3.0-noarch:core-3.1-amd64:core-3.1-noarch:core-3.2-amd64:core-3.2-noarch:core-4.0-amd64:core-4.0-noarch:core-4.1-amd64:core-4.1-noarch
    Distributor ID:	Ubuntu
    Description:	Ubuntu 14.04.4 LTS
    Release:	14.04
    Codename:	trusty

Execution:

    $ battery
    [{Current:open /sys/class/power_supply/BAT0/energy_now: no such file or directory}]

KenjiTakahashi, 2016-05-14:
Oh, it seems your battery driver returns data in Amperes. That's the less tested path, as I don't have any HW that does it.
Anyway, I think I got it, can you please check again?

TheHippo, 2016-05-15:
Seems to be fixed.

closed 2016-05-15 by TheHippo