distatus/battery #8: CLI doesn't work on Manjaro Linux?

PicoSushi, 2018-12-11:
I ran `go get -u github.com/distatus/battery/cmd/battery` from README and ran`battery` command, and I've got output as below.
```
[{Current:open /sys/class/power_supply/hidpp_battery_0/voltage_now: no such file or directory Full:open /sys/class/power_supply/hidpp_battery_0/voltage_now: no such file or directory Design:open /sys/class/power_supply/hidpp_battery_0/voltage_min_design: no such file or directory ChargeRate:open /sys/class/power_supply/hidpp_battery_0/voltage_now: no such file or directory Voltage:open /sys/class/power_supply/hidpp_battery_0/voltage_now: no such file or directory DesignVoltage:open /sys/class/power_supply/hidpp_battery_0/voltage_min_design: no such file or directory}]
```

`uname -a`
```
Linux kaguya 4.14.83-1-MANJARO #1 SMP PREEMPT Fri Nov 23 18:20:11 UTC 2018 x86_64 GNU/Linux
```

`ls -1  /sys/class/power_supply/`
```
AC@
BAT1@
hidpp_battery_0@
```
`ls -1  /sys/class/power_supply/hidpp_battery_0/ `
```
capacity_level
device@
manufacturer
model_name
online
power/
powers@
scope
serial_number
status
subsystem@
type
uevent
```

KenjiTakahashi, 2018-12-11:
Manjaro is probably not to able here, looks more like the laptop (or whatever) is reporting a bogus battery "device".
Can you please show the outputs of `cat /sys/class/power_supply/hidpp_battery_0/type` and `ls -1 /sys/class/power_supply/BAT1/`?

PicoSushi, 2018-12-12:
`cat /sys/class/power_supply/hidpp_battery_0/type`
```
cat: /sys/class/power_supply/hidpp_battery_0/type: No such file or directory
```

`ls -1 /sys/class/power_supply/BAT1/`

```
alarm
capacity
capacity_level
cycle_count
device@
energy_full
energy_full_design
energy_now
manufacturer
model_name
power/
power_now
present
serial_number
status
subsystem@
technology
type
uevent
voltage_min_design
voltage_now
```

external info: `acpi -V` works well.

```
attery 0: Discharging, 0%, rate information unavailable
Battery 1: Full, 100%
Battery 1: design capacity 2961 mAh, last full capacity 2726 mAh = 92%
Adapter 0: on-line
Thermal 0: ok, 45.0 degrees C
Thermal 0: trip point 0 switches to mode critical at temperature 105.0 degrees C
Thermal 1: ok, 45.0 degrees C
Thermal 1: trip point 0 switches to mode critical at temperature 130.0 degrees C
Cooling 0: Processor 0 of 3
Cooling 1: iwlwifi no state information available
Cooling 2: CHRG 0 of 4
Cooling 3: x86_pkg_temp no state information available
Cooling 4: Fan 1 of 1
Cooling 5: Processor 0 of 3
Cooling 6: Processor 0 of 3
Cooling 7: B0D4 no state information available
Cooling 8: TFN1 0 of 12
Cooling 9: INT3400 Thermal no state information available
Cooling 10: iwlwifi 0 of 19
Cooling 11: intel_powerclamp no state information available
Cooling 12: pch_skylake no state information available
Cooling 13: Processor 0 of 3

```

KenjiTakahashi, 2018-12-12:
> `cat /sys/class/power_supply/hidpp_battery_0/type`
> 
> ```
> cat: /sys/class/power_supply/hidpp_battery_0/type: No such file or directory
> ```
Now, that's strange. We can see in your earlier output that this file should exist.

Anyway, I think what's happening is that your device is exposing two battery outputs, one of which does not have actual battery information. It seems that the `hidpp` is some kind of Logitech HID extension. Do you happen to have a battery powered Logitech device connected?

The problem at hand is that the CLI does not have granular error reporting, if anything (i.e. getting info for that second device) fails, it just fails. This should be fairly easy to improve.

PicoSushi, 2018-12-19:
I doubt Solaar at first for creating `/sys/class/power_supply/hidpp_battery_*`, but read [source](https://github.com/pwr/Solaar) and found that was wrong. This might created by kernel? I'm not sure.

Anyway, this issue need to be solve and there's two options.
- Treat HID device battery as other battery.
    - It seems not HID battery dosen't battery have data as usual file (like `energy_now`), so implementing will require more hard work than other.
- Ignore HID device.

KenjiTakahashi, 2019-03-31:
Sorry for this super long delay! Was moving to another country recently, and haven't had much time for OSS...

Anyway, the problem at hand was error handling in the CLI part. I have just pushed an update to it on master. This should now return _both_ error message for HIDPP device, and battery info for actual battery.
I have no way to test it for your specific scenario, though, so please let me know how it works. Thanks!

As for supporting the HIDPP info, well, there is nothing to support really in the "standard" battery interface (the only usable part is `status` and, maybe, `capacity_level`). There may be more information stored somewhere else, but that I do not know.

KenjiTakahashi, 2019-04-14:
Gonna go ahead and close this, as the root cause should be fixed now. Feel free to reach out, if there are any more problems!

closed 2019-04-14 by KenjiTakahashi