distatus/battery #25: Not Working – Possible Issue with Windows 11??

blumn2, 2021-10-31:
I've recently installed oh-my-posh on my laptop, and when creating a custom theme to include the battery charging status, it gives me the following error: ```{State:Unknown Current:0 Full:0 Design:0 ChargeRate:0 Voltage:0 DesignVoltage:0}```.

I did some digging and saw that oh-my-posh uses this library for getting those values, and so I downloaded Go installed the cmd utility on my laptop and tried running that, to see if it was an issue with oh-my-posh, but when I ran the `battery` command, it returned the following output `Error getting info for BAT0: {ChargeRate:Unknown value received}`.

I am running Windows 11 on this laptop, so there may be a compatibility issue there? 

If there is any more information I can provide to help debug this issue, please let me know. I do not personally have any experience with Go, but am willing to help in whatever way I can. 

Thanks!

Khaos66, 2021-12-10:
I'm having the same issue on win10. So a win11 corelation is unlikely.

When debugging this I get a value of `bs.Rate=-2147483648` and [this can't be converted to Float64](https://github.com/distatus/battery/blob/master/battery_windows.go#L290).

I'm trying to understand the rest of this and find a solution =D

Khaos66, 2021-12-10:
Looks like this is [by design](https://github.com/distatus/battery/blob/master/battery_windows.go#L92).
Windows returns a value of `BATTERY_UNKNOWN_RATE` 
The issue is that windows isn't returning a valid charge/discharge rate... 
I guess we're out of luck here

Khaos66, 2021-12-10:
Maybe I was wrong. 
```go
// There is something wrong with this constant, but
// it appears to work so far...
if num == -0x80000000 { // BATTERY_UNKNOWN_RATE
    return 0, errors.New("Unknown value received")
}
```

There is something wrong ;D 
@KenjiTakahashi why is the constant negative? Looking it up online showed it should be positive.

Khaos66, 2021-12-10:
I've tryed to read the charge rate with other tools. It looks like Windows is not able to supply the charge rate
[image]

It also gets a value of `-2147483648`
[image]

So I guess this library isn't the issue. We need to report this to oh-my-posh and ignore the error there.

KenjiTakahashi, 2021-12-28:
Thanks for the report and thanks @Khaos66 for the investigation!

I think this is what my comment in the code was supposed to mean: That what I experienced in real life was not in line with the docs.

Out of curiosity, would you be able to test it on different OS (e.g. some Linux LiveCD)? If you're consistently getting invalid rate value, it would mean either the driver is faulty (a Windows problem) or the controller is misbehaving (a hardware problem).

Khaos66, 2021-12-30:
@KenjiTakahashi Sure thing!

On Ubuntu I get this error when running the sample
```
Error getting info for BAT0: {ChargeRate:read /sys/class/power_supply/BAT0/current_now: no such device}
```
So it looks like the hardware just can't deliver that value.

I'm using a `HP Spectre` notebook by the way.

What is strange though, it shows up in the terminal:
```bash
ubuntu@ubuntu:/sys/class/power_supply/BAT0$ ls -l
total 0
-rw-r--r-- 1 root root 4096 Dec 30 06:44 alarm
-r--r--r-- 1 root root 4096 Dec 30  2021 capacity
-r--r--r-- 1 root root 4096 Dec 30 06:44 capacity_level
-r--r--r-- 1 root root 4096 Dec 30  2021 charge_full
-r--r--r-- 1 root root 4096 Dec 30  2021 charge_full_design
-r--r--r-- 1 root root 4096 Dec 30  2021 charge_now
-r--r--r-- 1 root root 4096 Dec 30  2021 current_now
-r--r--r-- 1 root root 4096 Dec 30 06:44 cycle_count
lrwxrwxrwx 1 root root    0 Dec 30  2021 device -> ../../../PNP0C0A:00
drwxr-xr-x 3 root root    0 Dec 30  2021 hwmon2
-r--r--r-- 1 root root 4096 Dec 30  2021 manufacturer
-r--r--r-- 1 root root 4096 Dec 30  2021 model_name
drwxr-xr-x 2 root root    0 Dec 30  2021 power
-r--r--r-- 1 root root 4096 Dec 30  2021 present
-r--r--r-- 1 root root 4096 Dec 30  2021 serial_number
-r--r--r-- 1 root root 4096 Dec 30  2021 status
lrwxrwxrwx 1 root root    0 Dec 30  2021 subsystem -> ../../../../../../class/power_supply
-r--r--r-- 1 root root 4096 Dec 30  2021 technology
-r--r--r-- 1 root root 4096 Dec 30  2021 type
-rw-r--r-- 1 root root 4096 Dec 30  2021 uevent
-r--r--r-- 1 root root 4096 Dec 30  2021 voltage_min_design
-r--r--r-- 1 root root 4096 Dec 30  2021 voltage_now
```

But when I try to read the value, I get the same error:
```
ubuntu@ubuntu:/sys/class/power_supply/BAT0$ cat current_now 
cat: current_now: No such device
```

So, I think we can close this issue as this is a hardware problem ;)

Khaos66, 2021-12-30:
@astridos2go I fixed this issue in oh-my-posh some days ago, so you just need to update ;)

KenjiTakahashi, 2021-12-30:
Thanks for checking out!
Files inside `/sys` are not "real" files, so the kernel probably exposes the hierarchy that's supposed to work and only when you try to read from the `current_now` file, realises that it cannot read the value.

I agree that there's nothing we can do to avoid the error in such case. The only way is to handle such errors somehow. Which is already possible in the library.

closed 2021-12-30 by KenjiTakahashi